appliedengdesign/vscode-gcode-syntax #1: Incorrect highlighting for  DO / END

proulxwz, 2019-02-04:
This code shows incorrect highlighting for DO/ENDDO
(********************************)
WHILE[1.EQ1.]DO1(MAIN*LOOP)
(********************************)
#3004=0
#28=0(LOOP*CHECKING)
WHILE[ABS[#[#5+7.]]LT0.1]DO2(WAIT*FOR*OPCODE)
G04X#[#31+10.]
#28=#28+1
IF[#28 LT 5000] GOTO300
#33=78
GOTO9000
N300
END2

[image]

You will see that the While statement's "DO2" highlights the "O2"  and the END2 Statement has the "D2" portion highlighted.

mikecentola, 2019-02-04:
Fixed this. Will release v0.1.1 here shortly.

mikecentola, 2019-02-04:
@proulxwz Just published v0.1.1 to the marketplace. Let me know if that fixes the issues. Thanks!

mikecentola, 2019-02-06:
Closing issue by EOB today unless its still not working

proulxwz, 2019-02-07:
Looks good now!  Thanks

mikecentola, 2019-02-07:
Great!

closed 2019-02-07 by mikecentola