appliedengdesign/vscode-gcode-syntax #3: Enhancement: Support for square brackets

proulxwz, 2019-02-04:
Please see in this screenshot how square brackets are used for complex variable assingment and for logic evaluation:

[image]

It would be useful to have the brackets/content highlighed and also when used in complex variable assignments clearly depicted as variables

mikecentola, 2019-02-04:
Do you think that the highlighting of a bracket should override any highlighting inside? Highlighting the brackets themselves should be easy.

shapesorter, 2019-08-30:
I prefer the full set of brackets to be stand out, in programs I tried to make it like this:
[image]

mikecentola, 2020-01-02:
Fixed in v0.2.0

closed 2020-01-02 by mikecentola