appliedengdesign/vscode-gcode-syntax #4: Support for coloring .eia files

shapesorter, 2019-08-30:
Is it possible for me to make .eia files use this scheme, I cant seem to add it to the list in a Json.

mikecentola, 2019-09-02:
I'll add it to the next hotfix :)

shapesorter, 2019-09-04:
Thank You, I like your coloring scheme for the most part, I am a bit out of my element offering help, but I will be using it for now.  Thank you for putting it out there for me to use, and getting back to me so fast.

mikecentola, 2019-09-09:
Absolutely! I'm trying to get back into the swing of things and do more development on this project :)

shapesorter, 2019-09-27:
So, I realized a few days ago that I can enable coloring to work on the .EIA files as you have it set up now.  I needed to uconfigure it properly.  I can associate any file type.  I should now tell myself to read the documentation.
[image]

mikecentola, 2020-01-02:
Fixed in v0.2.0

closed 2020-01-02 by mikecentola