appliedengdesign/vscode-gcode-syntax #6: Outline view not useful: shows a node for every line

Psychlist1972, 2020-01-11:
I would have expected the move/cutting operations to be grouped. Is this the expected output?

[image]

mikecentola, 2020-01-12:
It bases the tree nodes on the G00/G01/G02/G03 commands. Since you have a G1 on each line, it's technically working as intended. You don't need a G1 on lines following. I could add some kind of logic to determine if it should suppress some of that but its doing a linear move each time.

Psychlist1972, 2020-01-12:
Thanks. Not my gcode; it's a sample file that came with an inexpensive 3018 CNC router.

I'm not sure if any other CAM programs generate code with the modal command on each line.

mikecentola, 2020-01-12:
> Thanks. Not my gcode; it's a sample file that came with an inexpensive 3018 CNC router.
> 
> I'm not sure if any other CAM programs generate code with the modal command on each line.

I know in the CAM program I use (RhinoCAM) the post processor has options to “optimize” code which will output the Modal and then just X Y or Z changes. 

I’ll close this for now. Thanks

closed 2020-01-12 by mikecentola