appliedengdesign/vscode-gcode-syntax #8: G-Code Tree disappear from available views

BernardG, 2020-01-12:
It has been working, so I am not sure what I did, but now I am unable to display the tree view, which, to me, is very handy to have. How do I get it back?

[image]

mikecentola, 2020-01-12:
The extension is written to hide the tree view when you’re not viewing a gcode file. Maybe I’ll change it to clear it and disable refresh but not hide it. 

Is it missing even when you’re viewing a gcode file?

BernardG, 2020-01-12:
Yes, once it has been disappearing, it does not show again when getting back to the G-Code file, and it's not available in the  view list, as shown on the screen shot.

mikecentola, 2020-01-13:
> Yes, once it has been disappearing, it does not show again when getting back to the G-Code file, and it's not available in the view list, as shown on the screen shot.

Ok let me pull it up tomorrow and take a look at the activation code.

closed 2020-05-11 by stale[bot]