appliedengdesign/vscode-gcode-syntax #9: Markup issue with division symbol

CodeDabbler23, 2020-01-29:
There is an issue with the markup associated with block commenting.
Please see the attached photo.
Really love your extension for managing CNC code - nice work!

[image]

mikecentola, 2020-02-06:
Working on a hotfix for it right now. Thanks for the kind words!

mikecentola, 2020-02-06:
Just pushed v0.3.2 and removed that line. @CodeDabbler23 please confirm and I'll close the issue. Thanks!

CodeDabbler23, 2020-02-07:
Looks good - thanks!!
I have some other thoughts and ideas, what is the best way to communicate those to you?

mikecentola, 2020-02-07:
Keybase chat, Facebook, Twitter, or email me? Should be able to find me pretty easily ;)

closed 2020-02-07 by mikecentola