appliedengdesign/vscode-gcode-syntax #17: Line Numbering

shadowbane900, 2021-02-12:
add Line numbering/ renumbering function

mikecentola, 2021-02-15:
Can you describe a little more what you're looking for?

shadowbane900, 2021-02-15:
in software such as cimco edit and predator editor can add line numbers i.e. N100 N110 and re number them as needed

mikecentola, 2021-02-15:
ok shouldn't be too hard to get that working. I'll add it to 0.6 roadmap.

silopolis, 2021-05-19:
This one too would be nice to have ! :+1: 
Being able to act on whole program as well as selected blocks would be great.
Option to ignore blank lines and comment blocks (other specific blocks ?) would be nice.

mikecentola, 2021-05-26:
@silopolis Right now it acts on the whole file, but has the options to ignore blanks and comments in the settings. Shouldn't be too hard to add the ability to act on selected blocks.

silopolis, 2021-05-26:
Nice, thanks 👍🙏

mikecentola, 2021-11-30:
Line numbering added to v0.7

closed 2021-11-30 by mikecentola

silopolis, 2021-12-01:
Le mar. 30 nov. 2021 à 23:44, Mike Centola ***@***.***> a
écrit :

> Line numbering added to v0.7
>
Nice thanks 👍