appliedengdesign/vscode-gcode-syntax #18: "Command 'gcode.gcodeTree.refresh" not found"

InkRF, 2021-02-14:
In the past, the extension has worked fine for me, but up until recently when I had to write another program, the gcode tree would just appear blank. Clicking the refresh button gave me the "Command 'gcode.gcodeTree.refresh" not found" message. Uninstalling/installing it did not seem to do anything, so I'm curious as to what I can do to get the gcode tree to show all my commands.

Also, whenever i open a gcode file, in the output it just gets stuck on "loading tree view". 
Version is v0.4.1

[image]

InkRF, 2021-02-14:
Well, adding line numbering seemed fix the issue, now when clicking refresh it refreshed, neat!

closed 2021-02-14 by InkRF

mikecentola, 2021-02-15:
That is odd for sure. I don't think I've experienced that. I've got 0.5 coming out this week hopefully too.

mikecentola, 2021-02-15:
Figured it out. Had to fix an Enum in the code.

InkRF, 2021-02-15:
Huh, glad to know it was not just user error, felt bad spamming the issues if that was the case, glad to see you found the problem/solution.
Take care!