appliedengdesign/vscode-gcode-syntax #19: Support for Open VSX

florian-str, 2021-02-22:
It would be great if you could publish your plugin to Open VSX as well, as the [VSCoduim-Project](https://github.com/VSCodium/vscodium) moved to this repository as it's default source of extensions and there for you need workarounds to use your extension.

According to the [documentation](https://github.com/eclipse/openvsx/wiki/Publishing-Extensions), it seems to be pretty easy to do and would improve the usability and maybe the reach of the extension.

mikecentola, 2021-02-22:
That sounds like a good idea. I will look into this for sure!

mikecentola, 2021-03-11:
Just waiting for verification of the publisher ID and I should be able to publish it over there.

mikecentola, 2021-03-12:
All set! Now available on Open-VSX :)

[https://open-vsx.org/extension/appliedengdesign/vscode-gcode-syntax](https://open-vsx.org/extension/appliedengdesign/vscode-gcode-syntax)

closed 2021-03-12 by mikecentola