appliedengdesign/vscode-gcode-syntax #21: Add hover text description of G-Codes

BelKed, 2021-03-08:
**Suggest a Feature you'd like to see**
I suggest to add hover text description of G-Codes. Related to: https://github.com/Milys/NC-Gcode-VSCode/pull/9.

**Additional context**
When you hover G-Code, this should show short description and link to more info (source website).
Source website is: https://marlinfw.org/meta/gcode.

mikecentola, 2021-03-09:
Great idea. I'll see if I can swing it into 0.6

shapesorter, 2021-03-15:
If it would be possible to make this a configurable option, as many will not use Marlins G-code.  Fancy would be allowing links to documentation that is machine specific, or a file where we could build our own list.

BelKed, 2021-03-15:
Good suggestion. But that link should be link to API...

shapesorter, 2021-03-15:
I don't know how to do an API, but I understand the general idea, if there is help that I can give, as a CNC guy I would like to do it.  I do not know what that is, so when you see it say something to me.

mikecentola, 2021-03-15:
@shapesorter @BelKed  I decided to make my own NPM module for this. It is going to take JSON files that I wrote a schema for similar to Marlin. It'll have CNC Milling, CNC turning and 3D printing descriptions. I did it as a separate project so that people can contribute to the descriptions as well.

[gcode-reference](https://github.com/appliedengdesign/gcode-reference)

Also I'll put some logic into there to automagically convert to HTML, markdown, etc. Might make for a great overall community reference that isn't machine-specific.

mikecentola, 2021-03-15:
Could even build it out as a consumable API

BelKed, 2021-03-15:
Nice 😄

shapesorter, 2021-03-15:
Very

shapesorter, 2021-03-15:
Now all I have to to is figure out how to use it.  Any way you could screenshot where to add the g-code info?  I will be able to figure from there.

mikecentola, 2021-03-15:
@shapesorter I would LOVE for you guys to help out with the descrptions. Let me finish the schema and the basic code for the module and then I'll make an easy way for people to contribute to it.

BelKed, 2021-03-15:
I could help you, just write, when it's finished 😄

shapesorter, 2021-03-15:
Well, I would LOVE to help make it better.  Mostly I just use it as a coloring scheme, but the potential is great.

mikecentola, 2021-03-17:
@shapesorter @BelKed Made some progress drafting up a skeleton of the package. Have a look and let me know what you think. I wanted to at least get the JSON schema written and some skeleton files up there. Mainly check out the format of the JSON files, and let me know if that's enough information to store about each code. Thanks!

[gcode-reference](https://github.com/appliedengdesign/gcode-reference)

BelKed, 2021-03-18:
I think, there should be also optional field `link` pointing to the source...

mikecentola, 2021-03-18:
@BelKed Like pointing back to a web reference of the same thing? Or to a different website that has it listed?

BelKed, 2021-03-18:
To a different website that has it listed – with examples or something like that...
For example: `M00` will point to https://marlinfw.org/docs/gcode/G000-G001.html

mikecentola, 2021-04-15:
Oh boy!!

[image]

mikecentola, 2021-04-20:
Added to v0.6.0. Still need to work on more descriptions in [gcode-reference](https://github.com/appliedengdesign/gcode-reference

closed 2021-04-20 by mikecentola