appliedengdesign/vscode-gcode-syntax #23: .gcode Generated by Prusa Slicer Not Recognized

nverdonk, 2021-04-13:
**Describe the bug**
G-Code files produced by [PrusaSlicer](https://www.prusa3d.com/prusaslicer/) are not recognized or highlighted by extension.  They are .gcode file extensions.

**To Reproduce**
Steps to reproduce the behavior:

1. Generate G-Code file using PrusaSlicer
2. Save file to Desktop
3. Open file in VS Code
4. Confirm file is not recognized as G-Code

**Expected behavior**
Expect the file type to be recognized and highlighted properly.  VS code seems to recognize that it is a gcode language but extension does not (see screenshot below), 

**Screenshots**
[image]

**Environment / Version:**

- OS Version: Windows 10
- VSCode Version: 1.55.2
- Extension Version: v0.5.1

**Additional context**
I might be missing a step in the process here as the extension is working properly if I make a blank .gcode file and type some commands in.
I have also copied all the text from the PrusaSlicer generated file and pasted into the user-made file and correct highlighting occurs

Let me know if this should be moved, tagged, or deleted.

mikecentola, 2021-04-14:
@nverdonk can you attach the file here so I can try it on my computer with the current dev version?

nverdonk, 2021-04-14:
@mikecentola Yes, here is a file to play around with.  I just sliced it now and confirmed the same issue is occurring.

Had to ZIP it as GitHub won't allow native upload of .gcode
[Drive Gear.zip](https://github.com/appliedengdesign/vscode-gcode-syntax/files/6311804/Drive.Gear.zip)

mikecentola, 2021-04-14:
@nverdonk I'm on it!

BelKed, 2021-04-14:
Visual Studio Code does not colorise that big files...
[image]
You can enable it by clicking on "Forcefully Enable Features" or in settings – `editor.largeFileOptimizations`

mikecentola, 2021-04-14:
@nverdonk yup @belkid is correct. I got the same error

[image]

nverdonk, 2021-04-14:
### Highlighting Fixed
Toggling `editor.largeFileOptimizations` corrected the highlighting issue, but the G-Code Tree from the extension still claimed to not recognize the G-Code file.

This time, however, a loading bar showed up; perhaps the file is too large for the tree as well?
[image]

mikecentola, 2021-04-14:
@nverdonk Could be because the file is so large. Let me do some investigating on that side as well.

nverdonk, 2021-04-14:
@mikecentola @BelKed Thank you for such quick responses and help!

@mikecentola I believe the file size is the issue with the tree.  I deleted pretty much the whole file (~550,000 lines to 374 lines) and the tree loads correctly now.
[image]

The tree would be so cumbersome for a file of that size anyway; don't think this will ruin anyone's workflow :), just wanted to make sure it wasn't a bug.

mikecentola, 2021-04-14:
No worries. I wonder if I should add a detection for large files to not even try if it's over a certain size.

nverdonk, 2021-04-14:
I think adding note or warning along the lines of

_This file exceeds maximum readable file size of _____ MB.  VS Code might disable highlighting and GCODE tree will not generate_

would help catch people in my situation in the future.

mikecentola, 2021-04-15:
@nverdonk yeah for sure. I'll see if I can find somewhere to add it. I'm working on the hover code right now.

mikecentola, 2021-04-19:
I added a warning popup for anything over 10K lines when autoRefresh is enabled.

nverdonk, 2021-04-19:
I think that's a great addition and will save people a lot of headache in the future!  I'll close out the issue for now and we can reopen if needed.

closed 2021-04-19 by nverdonk