appliedengdesign/vscode-gcode-syntax #24: Support for .g files.

Festivejelly, 2021-04-19:
## Support for .g files

**Additional context**
The plug doesnt seem to support .g files. Firmware such as reprapfirmware use .g files  for the configs of the duet boards. Would be handy to have it recognise this as gcode.

mikecentola, 2021-04-19:
@Festivejelly It's added to the 0.6 that I'll be releasing tonight or tomorrow.

@shapesorter It's hardcoded into the `package.json` file:

```json
"extensions": [
                    ".001",
                    ".apt",
                    ".aptcl",
                    ".cls",
                    ".cnc",
                    ".din",
                    ".dnc",
                    ".ecs",
                    ".eia",
                    ".fan",
                    ".fgc",
                    ".fnc",
                    ".g",
                    ".g00",
                    ".gc",
                    ".gcd",
                    ".gcode",
                    ".gp",
                    ".hnc",
                    ".knc",
                    ".lib",
                    ".m",
                    ".min",
                    ".mpf",
                    ".mpr",
                    ".msb",
                    ".nc",
                    ".ncc",
                    ".ncd",
                    ".ncf",
                    ".ncg",
                    ".nci",
                    ".ncp",
                    ".ngc",
                    ".out",
                    ".pim",
                    ".pit",
                    ".plt",
                    ".ply",
                    ".prg",
                    ".pu1",
                    ".rol",
                    ".sbp",
                    ".spf",
                    ".ssb",
                    ".sub",
                    ".tap",
                    ".xpi"
                ],
```

mikecentola, 2021-04-20:
Added in v0.6.0

closed 2021-04-20 by mikecentola