appliedengdesign/vscode-gcode-syntax #26: My regexes

ganzuul, 2021-05-04:
Hi,
I noticed that coordinates X10.01 lose highlight-colors after the dot, so here are the regexes I have been using:

```
"coords": {
    "patterns": [
        {
            "match": "(?i)([\\,]?[XUIYVJZWKACR][ \\t]*[\\-\\+]?[0-9]++(?!\\.))",
            "name": "invalid.gcode"
        },
        {
            "match": "(?i)[XUI]?[ \\t]*[\\-\\+]?[0-9\\.]+",
            "name": "axis.x.gcode"
        },
        {
            "match": "(?i)[ZWK]?[ \\t]*[\\-\\+]?[0-9\\.]+",
            "name": "axis.z.gcode"
        },
        {
            "match": "(?i)[CH]?[ \\t]*[\\-\\+]?[0-9\\.]+",
            "name": "axis.c.gcode"
        }
    ]
},
```

invalid.gcode warns about coordinates without a decimal point like "X100". It works together with settings.json in the .vscode folder of the workspace.

```
"editor.tokenColorCustomizations": {
	"textMateRules": [
		{
			"name":"invalid",
			"scope": "invalid.gcode",
			"settings": {
				"foreground": "#F00",
				"fontStyle": "underline",
			}
		},
		]
	}
```
I hope these are useful to the project!

mikecentola, 2021-05-07:
Thanks! I'll get it into an update

mikecentola, 2021-05-07:
Fixed in v0.6.2

closed 2021-05-07 by mikecentola