appliedengdesign/vscode-gcode-syntax #28: Add support for Siemens Sinumerik codes

silopolis, 2021-05-19:
First thank you veeeery much for this excellent extension !

As I'm working on a Siemens Sinumerik (840d) driven machine, I'd love to see this NC's specific codes better supported for ".mpf" and ".spf" files, mainly for syntax highlighting.

For example blocks like:
* WORKPIECE(,,,,,,,,): for stock definition
* MSG(""): for operator messages
* T="TOOL_BY_NAME": for tool names
* TRANS/ATRANS, ROT/AROT...: for transformations
* CYCLExxx(): for positionning and machining cycles
* MCALL: for modal sub-program calling
* ...

WIth a little guidance, I'd gladly help for this.

Thanks again. Keep the good stuff coming

mikecentola, 2021-05-27:
I'll add this to the feature list, but probably won't get to it in 0.7.X as I think it'll require a language server

patcon, 2021-06-22:
Also, really grateful to find this project! 🎉 And eager to meet someone else on GitHub who's using a sinumerik 840D, @silopolis 👋 

For reference, some other projects doing syntax highlighting for sinumerik flavour g-code:
- https://github.com/staniska/sinumerik-highlight (atom editor)
- https://github.com/deathaxe/sublime-s840d (sublimetext3 editor)

darkross67, 2022-02-10:
@mikecentola, first of all thank you so much for creating this amazing extension !
 
It's been 5 years since I'm working as a CNC Engineer and programming for different CN controller, such as Heidenhain, NUM1060, FANUC and Siemens (840d), via CAD/CAM System like CATIA or PTC Creo.

I'm working a lot with NUM1060 language, and it really looks like GCode. 

How can I do if I want to add this language, I don't know how to start and I have few questions in mind : 

-  Do you prefer to stay for one specific CN Language ? and then maybe copy this project for another language, if so tell me more about the Licence, I really want to be clean about that.

-  If no, adding specific language features inside your code, but it kinds of bugs me because all the files/logo reference 'G-Code' and it will no longer be the case after that, it will be more 'CNC-Code' or something, if you know what I mean.

I will really appreciate your feedback after that and guide me how I can support.

PS: Sorry for the mistakes, I'm French...