appliedengdesign/vscode-gcode-syntax #29: Highlighting inconsistency with macro variables & control functions

zachallaun, 2021-06-16:
First off, thanks for all the work that's been put into this! Really happy with it overall.

I ran into a couple highlighting issues while working on a macro. Please see the screenshot below. Things that seem "off" in the screenshot:

* `DO` and `GOTO` are being highlighted inconsistently (this seems to be the case on their own line as well)
* The last closing `]` on the `IF` line does not match the opening bracket in the same way that it does on the `WHILE` line

[image]

mikecentola, 2021-08-05:
Probably happened when I tweaked the Regex's for the coloring on the last update. I'll try to get it fixed up in the next update!

mikecentola, 2021-11-30:
Fixed in v0.7.0

closed 2021-11-30 by mikecentola