appliedengdesign/vscode-gcode-syntax #31: Wrong "Tool Changes" count

7enso, 2021-08-23:
"M6" in comments is counted as tool change. This is problematic in some code cause "M6" also means metric 6 mm thread - for example in the description of a tool.

Leads to "Tool Changes" count being wrong.

mikecentola, 2021-09-21:
Do you mean using M6 in a comment? I'm unfamiliar with where M6 would be used as an M-Code to indicate thread pitch.

7enso, 2021-09-28:
Yes, I mean just using "M6" somewhere in the comments is counted as a tool change so the count gets wrong. Using M6 not as an M-Code but for naming a tool in the comments for example. M6 in the whole world outside the US means "Metric 6 mm thread" and is used often in naming thread milling tools.

https://en.wikipedia.org/wiki/ISO_metric_screw_thread

Maybe you can somehow ignore the patterns used in comments for the tool change count?

mikecentola, 2021-11-11:
@7enso yeah good call. I'll see if I can have it ignore anything in comments when it's calculating.

mikecentola, 2021-11-30:
Just fixed this in the dev code, so it'll be in the next release!

closed 2021-11-30 by mikecentola