appliedengdesign/vscode-gcode-syntax #32: Some features are not working

laptop-laptop, 2021-09-11:
**Describe the bug**

Hi, first time user. Maybe i don't know how to use the extension properly.

After opening a gcode file in Visual Studio Code, although the syntax are colour coded, there is nothing shown under TREE. Do I need to do something like compiling or running? Since it is not a program for the PC, it may not work.

As for Hovers, sometimes it works but sometimes it doesn't. For example, moving the cursor over G90 shows "G90: Absolute Positioning". However, moving the cursor over M73, M862, M299, M292, M140, etc. shows nothing.  Moving over G1 just shows "G1:"

**Environment / Version:**

- OS Version: Windows 10
- VSCode Version: 1.60.0
- Extension Version: 0.6.2 

**Additional context**
Add any other context about the problem here.

laptop-laptop, 2021-09-11:
Not sure if this is related. I noticed that at the bottom right, I often get "File size is above 10K lines. Three / Stats refresh may not work." Can such limitation be lifted?

mikecentola, 2021-09-21:
The file size limit is there because it would take a long time to compute it. I can add a setting in there for it or a way to manually calculate the tree.

laptop-laptop, 2021-09-24:
Yes, please do that. I am not familiar with Github. Will I be notified once it is added? When shall I check back?

mikecentola, 2021-11-11:
@laptop-laptop I will be working on this shortly here. You should get a notification when it updates and I'll let you know on here. Thanks!

mikecentola, 2021-11-30:
Added the max filesize as user customizable in v0.7

closed 2021-11-30 by mikecentola