appliedengdesign/vscode-gcode-syntax #34: Exception in the debug console: `rejected promise not handled within 1 second: TypeError: Cannot read property 'substr' of undefined

CoderAllan, 2021-11-12:
**Describe the bug**
In Visual Studio Code I'm currently developing an extension for processing GCode files. 
When the extension is launched a new instance of VSCode opens where two gcode files are opened by default.
The moment I click on the gcode file that is not active the exception shown below is shown in the host VSCode instance debug console. See attached screenshot.

**Expected behavior**
The extension should be able to highlight the GCode without throwing the errors in the debug console.

**Environment / Version:**
- OS Version: Windows 10
- VSCode Version: 1.62
- Extension Version: 0.6.2

**Additional context**
Stacktrace from the debug console:
`rejected promise not handled within 1 second: TypeError: Cannot read property 'substr' of undefined
stack trace: TypeError: Cannot read property 'substr' of undefined
	at x.parseLine (c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.6.2\dist\extension.js:1:17709)
	at x.getBlocks (c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.6.2\dist\extension.js:1:12527)
	at new x (c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.6.2\dist\extension.js:1:12336)
	at E.parseTree (c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.6.2\dist\extension.js:1:22213)
	at E.refresh (c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.6.2\dist\extension.js:1:22010)
	at E.onDocumentChanged (c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.6.2\dist\extension.js:1:21198)
	at c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.6.2\dist\extension.js:1:9526
	at l.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:66:1712)
	at I.$acceptModelChanged (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:94:152916)
	at n._doInvokeHandler (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:13802)
	at n._invokeHandler (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:13486)
	at n._receiveRequest (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:12148)
	at n._receiveOneMessage (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:10826)
	at c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:8922
	at l.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:66:1712)
	at s.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:74:15990)
	at c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:113:30625
	at l.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:66:1712)
	at s.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:74:15990)
	at f._receiveMessage (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:74:21260)
	at c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:74:18138
	at l.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:66:1712)
	at w.acceptChunk (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:74:13351)
	at c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:74:12699
	at Socket.m (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:113:13575)
	at Socket.emit (events.js:315:20)
	at addChunk (internal/streams/readable.js:309:12)
	at readableAddChunk (internal/streams/readable.js:284:9)
	at Socket.push (internal/streams/readable.js:223:10)
	at Pipe.onStreamRead (internal/stream_base_commons.js:188:23)
`

**Screenshot**
[image]

mikecentola, 2021-11-30:
I can't seem to reproduce this on my end. I just updated the extension with the latest vscode engines. Give the new version a try and let me know if that helps it.

CoderAllan, 2021-12-01:
The extension has been updated to v0.7.0, but I still experience the same issue as you can see below. 
But I noticed that the exception is only thrown for two of the files I'm testing with.
So the attached files Test1 and Test2 are both producing the error, but the Test3 is not producing the error.
[Test1.txt](https://github.com/appliedengdesign/vscode-gcode-syntax/files/7630854/Test1.txt)
[Test2.txt](https://github.com/appliedengdesign/vscode-gcode-syntax/files/7630852/Test2.txt)
[Test3.txt](https://github.com/appliedengdesign/vscode-gcode-syntax/files/7630853/Test3.txt)

`rejected promise not handled within 1 second: TypeError: Cannot read property 'substr' of undefined
stack trace: TypeError: Cannot read property 'substr' of undefined
	at _.parseLine (c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.7.0\dist\extension.js:1:13392)
	at _.getBlocks (c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.7.0\dist\extension.js:1:8210)
	at new _ (c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.7.0\dist\extension.js:1:8019)
	at P.parseTree (c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.7.0\dist\extension.js:1:19392)
	at P.refresh (c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.7.0\dist\extension.js:1:19189)
	at P.onActiveEditorChanged (c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.7.0\dist\extension.js:1:17845)
	at c:\Users\Allan\.vscode\extensions\appliedengdesign.vscode-gcode-syntax-0.7.0\dist\extension.js:1:5128
	at l.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:66:1712)
	at c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:95:63131
	at l.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:66:1712)
	at b.acceptDocumentsAndEditorsDelta (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:94:156060)
	at b.$acceptDocumentsAndEditorsDelta (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:94:154149)
	at n._doInvokeHandler (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:13802)
	at n._invokeHandler (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\
resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:13486)
	at n._receiveRequest (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:12148)
	at n._receiveOneMessage (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:10826)
	at c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:8922
	at l.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:66:1712)
	at s.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:74:15990)
	at c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:113:30625
	at l.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:66:1712)
	at s.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:74:15990)
	at f._receiveMessage (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:74:21260)
	at c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:74:18138
	at l.fire (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:66:1712)
	at w.acceptChunk (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:74:13351)
	at c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:74:12699
	at Socket.m (c:\Users\Allan\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:113:13575)
	at Socket.emit (events.js:315:20)
	at addChunk (internal/streams/readable.js:309:12)
	at readableAddChunk (internal/streams/readable.js:284:9)
	at Socket.push (internal/streams/readable.js:223:10)
	at Pipe.onStreamRead (internal/stream_base_commons.js:188:23)
`

mikecentola, 2021-12-02:
@CoderAllan OK let me download the files and try them here. 

Side note, what kind of extension are your trying to make? Maybe we can incorporate it into this one?

mikecentola, 2021-12-02:
@CoderAllan figured it out. Working on a fix

CoderAllan, 2021-12-03:
> 
> Side note, what kind of extension are your trying to make? Maybe we can incorporate it into this one?

The extension I'm working on is a preview function for gcode.

mikecentola, 2021-12-04:
> > Side note, what kind of extension are your trying to make? Maybe we can incorporate it into this one?
> 
> The extension I'm working on is a preview function for gcode.

Oh awesome? Like backplotting? I've been working on trying to integrate backplotting into this one. We should collaborate for sure.

CoderAllan, 2021-12-04:
Yes, like backplotting. 2D only because my focus is laser engraving. I have no idea how to even start doing anything in 3D in vscode.
I will make a fork of vscode-gcode-syntax and dump what I got into it so you can see it. But be warned, there is a lot of work to be done before this could be released and the code is very proof-of-concept-messy at the moment.

mikecentola, 2021-12-04:
@CoderAllan awesome! I'll check it out for sure.

mikecentola, 2022-05-17:
Fixed in v0.7.1

closed 2022-05-17 by mikecentola