appliedengdesign/vscode-gcode-syntax #40: Highlighting not working properly

GotAnAccount, 2022-06-19:
**Describe the bug**
Highlighting is applied on letters only in whole words for a LinuxCNC g-code file. 

For example:
    o<gotots> sub
Both letters "t" are highlited blue and should not.

**To Reproduce**
Steps to reproduce the behavior:

1. Simply open a g-code file and view.
 
**Expected behavior**
No highting should normally happen for these words.

**Screenshots**
[image]

**Environment / Version:**
Version : 1.68.1
Validation : 30d9c6cd9483b2cc586687151bcbcd635f373630
Date : 2022-06-14T12:48:58.283Z
Electron : 17.4.7
Chromium: 98.0.4758.141
Node.js : 16.13.0
V8 : 9.8.177.13-electron.0
Système d’exploitation : Windows_NT x64 10.0.19044

**Additional context**
g-code file from this site: https://www.projectsbybrian.tech/s/PrintNC.zip
open file from "macros" folder.

mikecentola, 2022-06-21:
I'll look into this ASAP. Thanks!

mikecentola, 2022-07-01:
Fixed in v0.7.2

GotAnAccount, 2022-07-01:
Hi,

Thank you for the changes. I updated it (v0.7.2) and I can still see some use cases where the problem still seems to exist.

Here is the screenshot for the "while" and "endsub" cases:
[image]

Here is the screenshot for the "if" and "gt 0" and another "endsub":
[image]

mikecentola, 2022-07-10:
Okay I'll push changes for the t/f/b regex tonight or tomorrow

mikecentola, 2022-07-13:
@GotAnAccount Just pushed v0.7.3 with a bunch of fixes that should make editing the LinuxCNC ngc templates much nicer and more colorful :)

Let me know if it's working and we can close this issue out!

GotAnAccount, 2022-07-13:
OK, confirmed the changes did fix the issue. 

Thanks Mike for your excellent work!

closed 2022-07-13 by GotAnAccount