appliedengdesign/vscode-gcode-syntax #43: auto update not working

sslupsky, 2022-08-24:
**Describe the bug**
I am not sure if you are aware but the latest version showing in the VSCode is 0.7.0 but I see there have been a few updates since then and the current version is 0.7.4.

sslupsky, 2022-08-24:
Here is a screenshot.
[image]

mikecentola, 2022-08-26:
@sslupsky what version is your VSCode? v0.7.4 requires `^1.69.0`

I'm showing this in mine:

[image]

sslupsky, 2022-08-26:
I have tried it with both 1.69 and 1.70.  Currently on 1.69.2.

sslupsky, 2022-08-26:
Odd, today it appears to have updated somewhere along the way ... version shows 0.7.4 now.  How long does it typically take for an update to propagate?  Since I installed the plugin into a clean install of VSCode, I expected that 0.7.4 would have installed by default since no previous version of the extension was installed.  Also, I could not find any way to "force an update".

I'll close this issue.

closed 2022-08-26 by sslupsky

mikecentola, 2022-08-26:
@sslupsky That is odd for sure and I've never seen that. Keep an eye on it and let me know anything like that happens again.