appliedengdesign/vscode-gcode-syntax #47: Add or Remove Line Numbers Error following V0.7.5 Update

theGoldenLink, 2022-09-20:
Commands
 G-Code: G-Code: Remove Line Numbers 
 G-Code: G-Code Add Line Numbers 
These commands result in errors and will not add or remove line numbers.

Steps to reproduce the behavior:
With a g-code file open, (whether .txt, .nc)
1. Go to 'Command Palette.'
2. Click on 'G-Code: G-Code Add Line Numbers'
3. Error is made known by a pop-up presenting you with a phrase letting the user know the command did not work.

When the command is selected it should either remove or add line numbers.

- OS Version: Windows_NT x64 10.0.22000
- VSCode Version:1.71.2
- Extension Version: V0.7.5

On older Extension Version: V0.7.4 both commands work as expected.

mikecentola, 2022-09-21:
@theGoldenLink Can you let me know which settings you have enabled/disabled for the line numberer in configuration?

theGoldenLink, 2022-09-21:
Gcode > Line Numberer: Add Space After -- Enabled
Gcode > Line Numberer: Frequency -- Every Line (Default)
Gcode > Line Numberer: Ignore Blank -- Enabled
Gcode > Line Numberer: Ignore Comments -- Enabled
Gcode > Line Numberer: Ignore Program Numberer -- Enabled

mikecentola, 2022-09-21:
@theGoldenLink thanks. looking into it right now

mikecentola, 2022-09-21:
Ended up refactoring the entire line numberer with a bunch of new options and features as well. Tying up some loose ends with it right now and testing it out.

mikecentola, 2022-09-22:
Well it turns out that, it wasn't my line Numberer code at all. It's something to do with the way webpack bundles in production vs development. I couldn't reproduce the error on my machine when debugging, but when I opened another window with v0.7.5 installed, I could reproduce. At least the whole line numberer system is much better now and more suited to large files.

commit 306d364f (2022-09-23): Refactor registerCommands func into index. Fixes #47
closed 2022-09-23 by mikecentola