appliedengdesign/vscode-gcode-syntax #53: syntax highlighting error

alarak233, 2022-11-04:
hi everyone
The g-code with 'C' can't match more than 4 digits like this:

C8 X13.98488889 Z0.89909369
C8.2 X13.98488611 Z0.89920968
C8.4 X13.98488333 Z0.89930708
C8.6 X13.98488056 Z0.89938586
C8.8 X13.98487778 Z0.89944604
C9 X13.98487500 Z0.89948762
C9.2 X13.98487222 Z0.89951058
C9.4 X13.98486944 Z0.89951494
C9.6 X13.98486667 Z0.89950069
C9.8 X13.98486389 Z0.89946783
C10 X13.98486111 Z0.89941636
C10.2 X13.98485833 Z0.89934629
C10.4 X13.98485556 Z0.89925761
C10.6 X13.98485278 Z0.89915032
C10.8 X13.98485000 Z0.89902445
C11 X13.98484722 Z0.89887992
C11.2 X13.98484444 Z0.89871696
C11.4 X13.98484167 Z0.89853579
C11.6 X13.98483889 Z0.89837779
C11.8 X13.98483611 Z0.89837531
C12 X13.98483333 Z0.89855660
C12.2 X13.98483056 Z0.89873516
C12.4 X13.98482778 Z0.89889525
C12.6 X13.98482500 Z0.89903688
C12.8 X13.98482222 Z0.89915987
C13 X13.98481944 Z0.89926428
C13.2 X13.98481667 Z0.89935007
C13.4 X13.98481389 Z0.89941726
[image]

mikecentola, 2022-11-09:
@alarak233 curious... what model is this for? 3D printer or Mill/Lathe?

alarak233, 2022-11-10:
> 

Slow Tool Servo is used for the rapid manufacture of large departure freeform components on an Ultra Precision Diamond Turning Lathe.

mikecentola, 2022-11-10:
Very cool!

closed 2022-11-10 by mikecentola