appliedengdesign/vscode-gcode-syntax #56: Add support for RepRapFirmware meta G-code keywords

chrishamm, 2022-11-28:
First of all, thank you very much for this extension - it looks great and if I possible, I would really like to use the syntax highlighting from this project with Monaco in a web application to provide a nicer G-code editor.

What I've been missing is briefly mentioned in #44: Syntax highlighting for RepRapFirmware G-code keywords and expressions.

With RepRapFirmware it is possible to use variables (`var`, `global`, `set`), conditional code flow (`if`, `elif`, `else`, `while`, `abort`, `break`), to output generic messages (`echo`), and to use dynamic expression parameters (e.g. `G1 X{100 * 0.3}`, everything in curly braces is evaluated on execution). The set of keywords is relatively small and well-documented here: https://docs.duet3d.com/en/User_manual/Reference/Gcode_meta_commands

Is there any way this could be added to a future version or can you point me to the right bits that need to be changed?

mikecentola, 2022-12-09:
I think if I can find a way to change the syntax highlighting based on the machine type chosen in the configuration settings, then I can probably make this work! I'll put it on the schedule for v0.8, but I might have to push it to v0.9.

chrishamm, 2023-04-18:
The highlighting file works quite nicely, I may make some more minor changes before our next 3.5 release but it should be error-free.

mikecentola, 2023-04-25:
@chrishamm thanks for your work on that. I've been offline for quite some time now with other life issues, so I'm going to start playing catchup on this extension. I'll see if there is anything I can incorporate into this.

chrishamm, 2023-07-15:
There have been two minor changes in my source file since your last comment, just FYI.