appliedengdesign/vscode-gcode-syntax #58: Tree view does not show file commands

DeeEmm, 2023-01-18:
**Describe the bug**
Tree view does not display file commands even after manually refreshing. 

Runtime status also shows following error:

```
Cannot read properties of undefined (reading 'slice')
```

**To Reproduce**
Open file for viewing in VSCode. 

**Expected behaviour**
Tree view shows file commands

**Screenshots**

[image]

[image]

**Environment / Version:**

- OS Version: MacOS Ventura 13.1
- VSCode Version: Version: 1.74.3 (Universal)
- Extension Version: G-Code Syntax v0.7.7

**Additional context**
Syntax highlighting is okay, Manually refreshing the stats windows shows estimated runtime.

Some commands are not correctly highlighted in editor which may or may not be related 

For example globally setting feed rate...

```
F100
```

Or spindle speeds...

```
S500 M3
```

Also some Mcodes...

```
M101
M102
M103
```

mikecentola, 2023-04-25:
Interesting. Let me dive into this to see where there might be an issue. I have been slow getting back to this project because of some life issues.

DeeEmm, 2023-04-25:
All good I know how that gets.

If you need me to do any testing here just let me know. 

I'll add an example Gcode file when I get home this evening just in case it's syntax related.