appliedengdesign/vscode-gcode-syntax #69: Output Level settings being ignored - always having full output in output window

7enso, 2024-03-28:
In General:Output Level settings one can select different levels of Outputs.

When selecting "warnings" or "silent" I still get the full output in the output window. Since it's not possible to deactivate or hide the output window in VSCode it is taking up a lot of screen real estate every time I start VSCode.

Only workaround is minimizing the output window to two lines. If you completely close it it will be back to full size after VSCode restart.

mikecentola, 2025-09-05:
@7enso Hi thanks for the issue!  It sounds like there might be a bug in how I check the output level when logging to the output window. I will have to look into this. I know on my machine, the output window remains the same size no matter if it is outputting to it or not. Can you explain a little more about the functionality you are looking for?  If it worked correctly to output nothing or just warnings, that would work for you? Or are you looking for not generating the window itself at all? 

Thanks!

7enso, 2025-09-17:
@mikecentola Thanks for looking into this! I am looking for the following functionality:

when I select in Settings / "General:Output Level"

Full: I want evertyhing.
Warnings: I just wan't warnings.
Silent: I want nothing at all.